racerxdl/stm32f0-pico-dump #9: Does not work with Not locked STM32F3xx

CyberRSR, 2025-01-14:
I have successfully read locked and unlocked STM32F0. But when I try to read **not locked** STM32F3, I don't receive anything.
Does STM32F3xx have different SWD?

daohuaxiang4, 2025-01-18:
> 我已成功读取已锁定和已解锁的 STM32F0。但当我尝试读取**未锁定的**STM32F3 时，我什么也没收到。STM32F3xx 是否具有不同的 SWD？

I may be stupid, I hope you can teach me how to get the firmware of stm32f0 on Windows

Dilbert66, 2026-03-29:
The stm32f1 and f3 need to switch the swd pins to use swd instead of jtag which it defaults to.  That is done by sending 0xe79e (lsb first) after the inital reset and then send the 50 clock reset again.   It willl then read the non-locked devices fine after that.

Now having said that, the locked stm32f1xxx I have cannot be read anyhow as I don't think they suffer from this race condition.