racerxdl/stm32f0-pico-dump #11: Using Mosfet for external target powering leads to timeout.

JimKnopfIoT, 2025-04-20:
I use an Atten ST862-D target board with a STM32F0 on it. I had to remove two AMS1117 LDO from target board to get it to work. The voltage from pico pin is just enough to power the target board. As soon as i use a IRML2502 Mosfet to power the target board with an external PSU, the signal pattern on the oscilloscope changes. Data and reset signal shift and lead to a timeout. I can't figure out how to solve the problem. I attach the wiring and signal shape from oscilloscope. Yellow is pwr pin, violett is reset pin, green is data, blue is clock signal. I'm getting Error 128.

[99% ]user@host ~/stm32f0-pico-dump# python ./dump.py dump-mem4.bin
Reading from /dev/ttyACM0 to file dump-mem4.bin
Received ready. Starting dump
Received Starting
Received 08000000: 20000a88
Received Error reading: 128
Received DONE

[image]

[image]
[image]
[image]
[image]
[image]
[image]

Any idea how to fix this?

JimKnopfIoT, 2025-04-22:
I found a solution by accident. It seems it has nothing to do with the code. When i power my 3,3V target board with only 2 V, The dump.py script works without issues. It works in the range of 1,975 V up tp 2,060 V. Below or above this voltage range leads to Error 128. 

[image]
[image]
[image]
[image]

closed 2025-04-22 by JimKnopfIoT

racerxdl, 2025-04-27:
Thats interesting. Maybe the voltage was dropping when I was testing (Since I used the GPIO). Will test that eventually. Thanks for the info! 💯